NOCIONS/letswave6 #168: p value and t value

fnedez, 2019-06-07:
Hello,

I did a T test value between two datasets.
Where are the p value and t value plotted in the table?
See the image uploaded

Thank you 
Barbara
[image]

EVale, 2019-06-09:
Dear Barbara,

each time you want to output the table you should change the index pop-up menu and select whether you want to output a table for p values or t values.
The Ymin column is giving you the lowest value. It should apply the same for the T val. You should also be able to obtain an average t or p val in the selected time window.

Hope this helps.

best

elia

fnedez, 2019-06-10:
Dear Elia
Thank you very much for your help. Ok, now it is clear for the ymin value of both T and p vals.
However, what do you mean with "to obtain an average"? Average between which values?

Kindly
Barbara