NOCIONS/letswave6 #171: Edit channels: Incompatible electrode coordinate file

LisaGoller, 2019-09-05:
I'm working with a 128 channel set up. As I want to remove artifacts, I'd like to make use of the IC topographies. This means I need to edit my electrode coordinates. 
Unfortunately, letswave6 core functions only include the "standard-10-20-Cap81.locs" file, which is not compatible with my set up.

Do you have a ".locs" file for the 10-5 system including 128 channels available?

Kind regards,
Lisa

LisaGoller, 2019-09-05:
Maybe you could also help with that, as all of you asked questions somewhat related to this topic?
@LenaStroh
@bleonefe
@DT83 

Thanks in advance!
Lisa

fedeAR24, 2019-09-26:
Hi Lisa,

1- You can go to Edit --> Electrodes --> Edit electrodes coordinates.
2- Then click on 'Select custom file with channel locations' and browse to the '..\letswave6-master\resources\electrodes\fieldtrip_standard_locations'
3- Within this folder, there are two standard channel locations from BESA. One file ('BEM_standard_1005.elc') has the nose along -X, while the other ('BEM_standard_1005_XYswap.elc') has the nose along +Y.

The 'electrodes' folder has other locations from other manufacturers (biosemi, EGI). If you have another cap with non-standard locations, you might have to search for those elsewhere.

I hope it helps,
Federico

LisaGoller, 2019-09-26:
Dear Federico,

Thank you so much for your response.

It works!

Kind regards,
Lisa

fedeAR24, 2019-09-26:
You're welcome 🙂

closed 2019-09-26 by fedeAR24