NOCIONS/letswave6 #175: Error in import MFF format EGI file

fngwng, 2020-02-26:
Hi,

I am starting to use letswave to analyse EGI data these days, however, I am facing some problems when import the dataset, here is the error I got:

Undefined function or variable 'ft_notification'.

Error in ft_error (line 39)
  ft_notification(varargin{:});

Error in ft_read_header (line 2489)
        ft_error('unsupported header format "%s"', headerformat);

Error in RLW_import_EGI (line 54)
        hdr=ft_read_header(filename);

Error in LW_import_EGI (line 88)
            [out_datasets(filepos).header out_datasets(filepos).data
            message_string]=RLW_import_EGI(filename,'concatenate',configuration.parameters.concatenate,'implementation',configuration.parameters.implementation);

Error in letswave_gui>menu_callback (line 387)
 
Error while evaluating Menu Callback.

Does anyone have any idea of it? Appreciated any input!!!

Thanks!

LisaGoller, 2020-02-29:
Hey there!

I had the same problem at first. The solution seems super trivial, but makes a huge difference:

Go to the vmrk file and replace the huge gap between the trigger code and the next variable with a comma (edit…fill in gap, replace with “,” and apply to all).

Save this file and import vhdr file into letswave again.

I hope this will help you!

--Lisa

fngwng, 2020-03-02:
Hi Lisa,

Thank you very much for your help! However, when I import data file, in letswave 6, there is a "File" button, and then I go to "Import signals", on the right side, it shows many options for different format signals, such as "Import ASCII text files", "Import Biosemi BDF files"....for the data that I collected through EGI system, I guess need to choose "Import Netstation EGI files".

As you said "Save this file and import vhdr file into letswave again." there seems no place for importing vhdr file? Also, I couldn't find the vmrk file in the MFF folder...

Any idea about this, appreciate your help, thanks a lot!

LisaGoller, 2020-03-03:
Hello again☺

So, first things first:
Usually, when I make an EEG recording, I get three output files; an EEG file, a VHDR file, and a VMRK file.

The VMRK is the one where I had to make the adjustments I told you about in our last email (replace gap with comma after trigger code).

Once this was done, I could import them into letswave via “Brainvision VHDR files” (https://github.com/NOCIONS/letswave6/wiki/Import-signals), which can take a while depending on the duration of your recording.

I’m sorry if this doesn’t help you…I can only speak from how this worked with my own data files;)

Have a lovely day,
Lisa