NOCIONS/letswave6 #179: Error in plotting scalpmap timeseries (with assigned channel location)

ValeTheIstrice, 2020-07-02:
After channel.location assignment, it gives me this error when I try to plot the scalp-map:

_"Index exceeds the number of array elements (54).

Error in CLW_headplot (line 42)
vector=vector(find(tp==1));

Error in GLW_figure_scalpmap_series>headplot_btn_Callback (line 594)
        CLW_headplot(vector2,dataset.header,'maplimits',[cmin cmax]);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_figure_scalpmap_series (line 24)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_figure_scalpmap_series('headplot_btn_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback._"

The number of elements in the array seems to be 2 less than the actual number of electrodes, my montage is 58 but in the first attempt the error said "Index exceeds the number of array elements (56)" thus I firts tried to remove two electrodes from my dataset passing from 58 to 56, but now it gives me the same error (that you see above) and the array is 54 while the channels are 56.

amouraux, 2020-07-02:
To solve this, I would need the dataset or an example dataset that reproduces the error.

Best wishes,
André

ValeTheIstrice, 2020-07-02:
Thanks,
this is one of the dataset.

it's the grand average of 24 participants.

-- 
Valentina Nicolardi, Ph.D

*Social **Cognitive **and Affective Neuroscience Lab*
Department of Psychology.
University of Rome "La Sapienza".
Via dei Marsi 78 - 00185 - Roma.
Phone: (+39) 06-49917635. Fax: (+39) 06-49917635
 *e-mail: valentina.nicolardi@uniroma1.it <vanessa.era@uniroma1.it>*
https://agliotilab.org/lab-staff/senior-fellows/valentina-nicolardi#anchor

Scientific collaborator
[image: 1538123418803_PastedImage]
Institute of neuroscience IONS
Avenue Mounier, 53 bte B1.53.04 - 1200 Brussels

amouraux, 2020-07-02:
Missing attachment? ;-)
André

--
Valentina Nicolardi, Ph.D

*Social **Cognitive **and Affective Neuroscience Lab*
Department of Psychology.
University of Rome "La Sapienza".
Via dei Marsi 78 - 00185 - Roma.
Phone: (+39) 06-49917635. Fax: (+39) 06-49917635
*e-mail: valentina.nicolardi@uniroma1.it<mailto:valentina.nicolardi@uniroma1.it> <vanessa.era@uniroma1.it<mailto:vanessa.era@uniroma1.it>>*
https://agliotilab.org/lab-staff/senior-fellows/valentina-nicolardi#anchor

Scientific collaborator
[image: 1538123418803_PastedImage]
Institute of neuroscience IONS
Avenue Mounier, 53 bte B1.53.04 - 1200 Brussels