NOCIONS/letswave6 #185: ERROR when importing BDF file in letswave 6

joycescheirlinckx, 2020-11-24:
Hi everyone, 
When I want to import a BDF data file into letswave, I get the following error from Matlab: 

Invalid MEX-file
'\\client\c$\Users\Joyce\Documents\SCHOOL\2020-2021\stage\letswave6-master\external\fieldtrip\ft_getopt.mexw64':
Access is denied.

Error in ft_read_data (line 115)
hdr             = ft_getopt(varargin, 'header');

Error in RLW_import_BDF (line 26)
dat=ft_read_data(filename);

Error in LW_import_BDF (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data message_string]=RLW_import_BDF(filename);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating Menu Callback.

Does anyone know what this means and what I can do about it? 

Thanks!

amouraux, 2020-11-25:
Can you send me an example dataset?

Best wishes,
André

joycescheirlinckx, 2020-11-25:
[Pictogram bdf] S09.bdf<https://ugentbe-my.sharepoint.com/:u:/r/personal/emiel_cracco_ugent_be/Documents/Projects/Ongoing/Postdoc/Research%20Stay/Experiment/Data/S9/EEG/S09.bdf?csf=1&web=1&e=OyOX7p>

joycescheirlinckx, 2020-11-26:
Hi André,

In the meantime I found what the problem was 😊 But thank you anyway!

Best,
Joyce