NOCIONS/letswave6 #189: Can't generate table with multiple datasets with SS-EP data

Mavelasq, 2021-04-09:
Hi everyone, 
It is giving me the following error when I try to select multiple datasets within the "Generate table and graphs with SS-EP data" window:

Reference to non-existent field 'average_epochs_chk'.

Error in GLW_explore_SSEP>table_listbox_Callback (line 651)
       set(handles.average_epochs_chk,'Value',0);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in GLW_explore_SSEP (line 24)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)GLW_explore_SSEP('table_listbox_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.

It has no issue when generating a table for only one file, it only occurs with multiple. 

Thanks, 
Miguel

amouraux, 2021-04-09:
Dear Miguel,

I tried to fix the bug. Can you update and try again?

Best wishes,
André

Mavelasq, 2021-04-09:
Thank you André!
It is allowing me to select multiple datasets and generate table for each, but it opens separate figure windows per data set table. 

Best, 
Miguel