NOCIONS/letswave6 #196: Invalid MEX-file while importing L7 tutorial data

lepetitemin, 2022-02-01:
Dear Letswave developers and users, 

I am new to LW, just started the LW tutorial (https://letswave.cn/tu_ch1_1.html ) and facing this error "Invalid MEX-file.... \letswave7\external\fileio\private\ft_getopt.mex64': Access is denied." 

I have checked the reported Issues here and mathworks but couldn't find a solution for my situation. 

I have this problem with my uni computer which has MATLAB 2019b. I do not have the same issue on my personal laptop with MATLAB 2020b but this laptop doesn't have enough capacity to run all my data. 
The version of MATLAB wasn't the cause as I tried with MATLAB 2019b, 2020b, and 2021b. 
I also tried different ways of setting the path and checking the security for quarantined items, no luck.

Please share your solutions or any advice, I will greatly appreciate it!
Minnie

amouraux, 2022-02-01:
Dear Minnie,

Can you check whether the .mex64 file actually exists in this folder?

Best wishes,
André

lepetitemin, 2022-02-01:
Hi Andre, 
Thank you for asking! Yes, that was the first thing I did :'( 
In 2016, some people (including you) said updating the mexw64 file (#85 ) solved the issue, is it still be the case?  
If it is, could you please let me know how to?

Thanks, 
Minnie

lepetitemin, 2022-02-02:
If updating is replacing the file, I tried but didn't change a thing. But I also suspect it might be something to do with the virtual desktop set up so contracted IT support. 
Will keep you posted, meantime if you have any suggestions, please do.     

Cheers, 
Minnie