NOCIONS/letswave6 #198: Average participants

sarahldgn, 2022-12-09:
Hello, 
I'm very new to letswave (and to MatLab as well) and I'm doing frequency-domain analysis with data collected by frequency-tagging. I looked at every participants individually by averaging their epoch. However now I would like to average all the participants for every epoch but I can't figure out how to do it. Each time I get this message : 

"Array dimensions must match for binary array op.

Error in RLW_grand_average (line 60)
    out_data=out_data+(data*dataset_weights(setpos));

Error in LW_grand_average (line 76)
        [out_datasets(1).header,out_datasets(1).data,message_string]=RLW_grand_average(datasets,'dataset_weights',configuration.parameters.dataset_weights);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating Menu Callback."

Could someone help me with this?
Thanks!

Amaulu94, 2023-01-20:
Hello, 
I do have the same problem. Did you find a way to do it?
thanks

sarahldgn, 2023-01-24:
Hello, 
Yes I found a way but not the better way I think. Since I don't have too many participants, I juste averaged two participants, and then three, and then four, etc... And it seems to work. I hope it will help you.