NOCIONS/letswave6 #203: read_24bit issue while opening .bdf file

OswaldChr, 2024-06-21:
Hello, 

I get this problem while trying to open a .bdf file, and do not understand what it means : 

One or more output arguments not assigned during call to "read_24bit".

Error in read_biosemi_bdf>readLowLevel (line 282)
  buf = read_24bit(filename, offset, numwords);

Error in read_biosemi_bdf (line 257)
      buf = readLowLevel(filename, offset, epochlength*nchans); % see below in subfunction

Error in ft_read_data (line 408)
    dat = read_biosemi_bdf(filename, hdr, begsample, endsample, chanindx);

Error in RLW_import_BDF (line 26)
dat=ft_read_data(filename);

Error in LW_import_BDF (line 86)
            [out_datasets(filepos).header out_datasets(filepos).data message_string]=RLW_import_BDF(filename);

Error in letswave_gui>menu_callback (line 387)
                [configuration,datasets]=eval(eval_st);
 
Error while evaluating Menu Callback.

Gabrock94, 2024-11-21:
Hi! We are having the same issue as well. Did you @OswaldChr managed to find the reason?

amouraux, 2024-11-22:
> Vous n’obtenez pas souvent d’e-mail à partir de ***@***.*** Pourquoi c’est important<https://aka.ms/LearnAboutSenderIdentification>

Does this issue arise only with a specific file?
https://www.biosemi.nl/forum/viewtopic.php?t=1382
https://mailman.science.ru.nl/pipermail/fieldtrip/2019-August/039544.html

If so, could it be that the recording was interrupted abnormally?
Biosemi has a repair tool (ActiTools) that allows making a new BDF file from a file with a corrupted file header.

If the problem occurs for all BDF files, you might want to check which read_24bit function is being called using the which function.

Best wishes,
André

Gabrock94, 2024-11-25:
Hi @amouraux,
Thanks for the reply.

> If so, could it be that the recording was interrupted abnormally?
> Biosemi has a repair tool (ActiTools) that allows making a new BDF file from a file with a corrupted file header.

We had no luck with those tools and I am able to open it in LW on my linux machine. Most likely some funny stuff going on on Windows/Matlab. We will try with a different Matlab and version and see if it works.

amouraux, 2024-11-25:
Probably related to the fact that Matlab calls different mex files depending on the platform. Will check whether these files have been updated. Sorry for this!